t2linux/T2-Ubuntu #104: Unable to use firmware.sh on old macOS

irenatorosyan, 2023-11-28:
My computer is running macOS Mojave Version 10.14.6. Unfortunately, as I already partitioned my disk for Ubuntu, I am unable to update to later versions of macOS to be able to run the script. I am not sure if this is of any help, but I also have a mac on M2 chip running macOS Sonoma. Is there a way to use the arm computer or somehow run the script on older version of macOS?

lemmyg, 2023-11-28:
If you computer a new arm. You should try Asahi Linux.

AdityaGarg8, 2023-11-28:
> My computer is running macOS Mojave Version 10.14.6. Unfortunately, as I already partitioned my disk for Ubuntu, I am unable to update to later versions of macOS to be able to run the script. I am [...]

https://github.com/AdityaGarg8/Apple-Firmware/releases/download/Ventura/apple-firmware.deb

Install this deb and WiFi should start working.,

irenatorosyan, 2023-11-28:
> https://github.com/AdityaGarg8/Apple-Firmware/releases/download/Ventura/apple-firmware.deb
>
> Install this deb and WiFi should start working.,

Thank you for the suggestion! Just tried it out but I doesn't seem to work for me unfortunately.

irenatorosyan, 2023-11-28:
> If your computer a new arm. You should try https://asahilinux.org This project is for old Macs with T2 chip.

I have ubuntu on my old mac with T2. Just thought the new one may be helpful to fix the issues. Currently not trying to run Ubuntu on arm but I'll remember the advice for the future. Thank you!

AdityaGarg8, 2023-11-28:
> Thank you for the suggestion! Just tried it out but I doesn't seem to work for me unfortunately.

After installing this, what's 

1. `ls /lib/firmware/brcm`
2. `journalctl -k --grep=brcmfmac`

irenatorosyan, 2023-11-28:
> After installing this, what's
>
>     1. `ls /lib/firmware/brcm`
>
>     2. `journalctl -k --grep=brcmfmac`

Here's the output

brleude@brleude:~$ ls /lib/firmware/brcm
 BCM-0a5c-6410.hcd.zst
 BCM-0bb4-0306.hcd.zst
 bcm4329-fullmac-4.bin.zst
 bcm43xx-0.fw.zst
 bcm43xx_hdr-0.fw.zst
 brcmbt4377b3-apple,formosa.bin
 brcmbt4377b3-apple,formosa.ptb
 brcmbt4378b1-apple,atlantisb-m.bin
 brcmbt4378b1-apple,atlantisb-m.ptb
 brcmbt4378b1-apple,atlantisb-u.bin
 brcmbt4378b1-apple,atlantisb-u.ptb
 brcmbt4378b1-apple,capri-m.bin
 brcmbt4378b1-apple,capri-m.ptb
 brcmbt4378b1-apple,capri-u.bin
 brcmbt4378b1-apple,capri-u.ptb
 brcmbt4378b1-apple,honshu-m.bin
 brcmbt4378b1-apple,honshu-m.ptb
 brcmbt4378b1-apple,honshu-u.bin
 brcmbt4378b1-apple,honshu-u.ptb
 brcmbt4378b1-apple,santorini-m.bin
 brcmbt4378b1-apple,santorini-m.ptb
 brcmbt4378b1-apple,santorini-u.bin
 brcmbt4378b1-apple,santorini-u.ptb
 brcmbt4378b1-apple,shikoku-m.bin
 brcmbt4378b1-apple,shikoku-m.ptb
 brcmbt4378b1-apple,shikoku-u.bin
 brcmbt4378b1-apple,shikoku-u.ptb
 brcmbt4378b3-apple,kyushu-u.bin
 brcmbt4378b3-apple,kyushu-u.ptb
 brcmbt4387c2-apple,hokkaido-u.bin
 brcmbt4387c2-apple,hokkaido-u.ptb
 brcmbt4387c2-apple,madagascar-u.bin
 brcmbt4387c2-apple,madagascar-u.ptb
 brcmbt4387c2-apple,maldives-u.bin
 brcmbt4387c2-apple,maldives-u.ptb
 brcmbt4387c2-apple,okinawa-u.bin
 brcmbt4387c2-apple,okinawa-u.ptb
 brcmfmac43012-sdio.bin.zst
 brcmfmac43012-sdio.clm_blob.zst
 brcmfmac43143-sdio.bin.zst
 brcmfmac43143.bin.zst
 brcmfmac43236b.bin.zst
 brcmfmac43241b0-sdio.bin.zst
 brcmfmac43241b4-sdio.Advantech-MICA-071.txt.zst
'brcmfmac43241b4-sdio.Intel Corp.-VALLEYVIEW C0 PLATFORM.txt.zst'
 brcmfmac43241b4-sdio.bin.zst
 brcmfmac43241b5-sdio.bin.zst
 brcmfmac43242a.bin.zst
 brcmfmac4329-sdio.bin.zst
 brcmfmac4330-sdio.Prowise-PT301.txt.zst
 brcmfmac4330-sdio.bin.zst
 brcmfmac4334-sdio.bin.zst
'brcmfmac43340-sdio.ASUSTeK COMPUTER INC.-TF103CE.txt.zst'
 brcmfmac43340-sdio.bin.zst
 brcmfmac43340-sdio.meegopad-t08.txt.zst
 brcmfmac43340-sdio.pov-tab-p1006w-data.txt.zst
 brcmfmac43340-sdio.predia-basic.txt.zst
 brcmfmac4335-sdio.bin.zst
 brcmfmac43362-sdio.WC121.txt.zst
 brcmfmac43362-sdio.bin.zst
 brcmfmac43362-sdio.cubietech,cubietruck.txt.zst
[... 241 lines cut ...]
 brcmfmac4387c2-pcie.apple,okinawa.bin
 brcmfmac4387c2-pcie.apple,okinawa.clm_blob
 brcmfmac4387c2-pcie.apple,okinawa.txcap_blob
 brcmfmac4387c2-pcie.apple,okinawa.txt
 brcmfmac54591-pcie.clm_blob.zst
 brcmfmac89459-pcie.apple,hawaii-YSBC-m-2.3.txt
 brcmfmac89459-pcie.apple,hawaii-YSBC-m-2.5.txt
 brcmfmac89459-pcie.apple,hawaii-YSBC-u-4.1.txt
 brcmfmac89459-pcie.apple,hawaii-YSBC-u-4.3.txt
 brcmfmac89459-pcie.apple,hawaii.bin
 brcmfmac89459-pcie.apple,hawaii.clm_blob
 brcmfmac89459-pcie.apple,hawaii.txcap_blob
brleude@brleude:~$ journalctl -k --grep=brcmfmac
Nov 28 21:57:11 brleude kernel: usbcore: registered new interface driver brcmfm>
Nov 28 21:57:11 brleude kernel: brcmfmac 0000:02:00.0: enabling device (0000 ->>
Nov 28 21:57:11 brleude kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/br>
Nov 28 21:57:11 brleude kernel: brcmfmac 0000:02:00.0: Direct firmware load for>
Nov 28 21:57:11 brleude kernel: brcmfmac 0000:02:00.0: Direct firmware load for>
Nov 28 21:57:11 brleude kernel: brcmfmac 0000:02:00.0: Direct firmware load for>
Nov 28 21:57:11 brleude kernel: brcmfmac 0000:02:00.0: Direct firmware load for>
Nov 28 21:57:11 brleude kernel: brcmfmac 0000:02:00.0: Direct firmware load for>
Nov 28 21:57:11 brleude kernel: brcmfmac 0000:02:00.0: Direct firmware load for>
Nov 28 21:57:11 brleude kernel: brcmfmac 0000:02:00.0: Direct firmware load for>
Nov 28 21:57:11 brleude kernel: brcmfmac 0000:02:00.0: brcmf_pcie_setup: Dongle>
lines 1-11/11 (END)

AdityaGarg8, 2023-11-28:
You haven't sent the journalctl command properly

AdityaGarg8, 2023-11-28:
You can save it to a file as `journalctl -k --grep=brcmfmac > out.log` and copy and paste the text from there.

irenatorosyan, 2023-11-29:
Oh, I'm sorry, haven't noticed it's cut off

Nov 29 17:02:20 brleude kernel: usbcore: registered new interface driver brcmfmac
Nov 29 17:02:20 brleude kernel: brcmfmac 0000:02:00.0: enabling device (0000 -> 0002)
Nov 29 17:02:20 brleude kernel: brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4355c1-pcie for chip BCM4355/12
Nov 29 17:02:20 brleude kernel: brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac4355c1-pcie.apple,hawaii-YSBC-m-2.5-X0.bin failed with error -2
Nov 29 17:02:20 brleude kernel: brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac4355c1-pcie.apple,hawaii-YSBC-m-2.5.bin failed with error -2
Nov 29 17:02:20 brleude kernel: brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac4355c1-pcie.apple,hawaii-YSBC-m.bin failed with error -2
Nov 29 17:02:20 brleude kernel: brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac4355c1-pcie.apple,hawaii-YSBC.bin failed with error -2
Nov 29 17:02:20 brleude kernel: brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac4355c1-pcie.apple,hawaii-X0.bin failed with error -2
Nov 29 17:02:20 brleude kernel: brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac4355c1-pcie.apple,hawaii.bin failed with error -2
Nov 29 17:02:20 brleude kernel: brcmfmac 0000:02:00.0: Direct firmware load for brcm/brcmfmac4355c1-pcie.bin failed with error -2
Nov 29 17:02:20 brleude kernel: brcmfmac 0000:02:00.0: brcmf_pcie_setup: Dongle setup failed

AdityaGarg8, 2023-11-29:
Bluetooth on your model does not need firmware from macOS

AdityaGarg8, 2023-11-29:
Can you file a bug report to upstream kernel? The wifi driver is not made by us.

irenatorosyan, 2023-11-29:
I will open the issue there. Thank you for help!

closed 2023-11-29 by irenatorosyan