t2linux/T2-Ubuntu #106: Linux kernel v6.6.5-1 on mbp late 2019 causing crash

olefasting, 2023-12-10:
I had to revert to the previous kernel, as the machine just kept grinding to a halt as I opened the first applications (alacritty etc), right after logging in to the Gnome shell. The machine is not working excessively (or at all, really) as it hangs, rather it will just gradually go unresponsive.

I will not be able to dig any deeper into this today, but I have attached the journalctl output for the last two boots. Ping me if there is anything else I can do to assist.

[boot_minus_2.log](https://github.com/t2linux/T2-Ubuntu/files/13626996/boot_minus_2.log)
[boot_minus_1.log](https://github.com/t2linux/T2-Ubuntu/files/13626997/boot_minus_1.log)

AdityaGarg8, 2023-12-10:
I ain't able to reproduce your issue on My Mac, although I am using Ubuntu 22.04

I've set a kernel to compile. Can you test it @olefasting?

You need to download it from the artifacts from https://github.com/t2linux/T2-Ubuntu-Kernel/actions/runs/7157248798. 

Install the linux-headers deb first using `sudo apt install --reinstall /path/to/deb` and then linux-image the same way. Restart and report the results.

lemmyg, 2023-12-11:
Unfortunately, some issues also in Kubuntu 23.10.
I cant say what it is but the touchbar stops working and most snap application gets stuck.

In one of the boot I got this error:
Dec 11 08:28:32 kubuntuMac (udev-worker)[703]: 7-6: /etc/udev/rules.d/99-touchbar-tiny-dfr.rules:1 Failed to write ATTR{/sys/devices/pci0000:00/0000:00:1b.0/0000:04:00.1/apple-bce/apple-bce/bce-vhci/usb7/7-6/bConfigurationValue}, ignoring: Connection timed out
Could be that apple-bce has issues with the new version

lemmyg, 2023-12-11:
seems 6.6.6 got back to normal

closed 2024-03-10 by AdityaGarg8