t2linux/T2-Ubuntu #108: No wifi in patched ubuntu

moshin34, 2024-02-21:
1. Clean install of Mac OS
2. Ran the firmware.sh in MacOS
3. Installed the patched ubuntu 23.10 iso
4. Ran the firmware.sh in linux

When I then run sudo journalctl -k --grep=brcmfmac

I get 

before the lines that say direct firmware load failed with code -2, it says:

brcmf_fw_alloc_request: using brcm/brcmfmac43551c1-pcie for chip BMC4355/12. (in red)

and after the direct firmware load failed with code: -2's 

brcmf_pcie_setup: Dongle setup failed (in red)

When I manually run the commands I go to /tmp/apple-wifi-efi and firmware.sh isnt there.  

When I copy it there, or run from anywhere It just echoes

Detected Linux
Re-mounting the EFI Partition
Getting Wi-Fi and Bluetooth Firmware

I noticed there is a folder in /tmp/apple-wifi-fw but it seems empty.

AdityaGarg8, 2024-02-21:
```
echo "1" | sudo tee /sys/bus/pci/devices/0000:01:00.0/remove
sleep 1
echo "1" | sudo tee /sys/bus/pci/rescan
```

Try this

AdityaGarg8, 2024-02-21:
Also share `lspci`

moshin34, 2024-02-21:
lspci: 

00:00.0 Host bridge: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers (rev 02)
00:02.0 VGA compatible controller: Intel Corporation UHD Graphics 617 (rev 02)
00:14.0 USB controller: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller (rev 21)
00:16.0 Communication controller: Intel Corporation Sunrise Point-LP CSME HECI #1 (rev 21)
00:1c.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #1 (rev f1)
00:1c.4 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #5 (rev f1)
00:1d.0 PCI bridge: Intel Corporation Sunrise Point-LP PCI Express Root Port #9 (rev f1)
00:1e.0 Signal processing controller: Intel Corporation Sunrise Point-LP Serial IO UART Controller #0 (rev 21)
00:1f.0 ISA bridge: Intel Corporation Device 9d4b (rev 21)
00:1f.2 Memory controller: Intel Corporation Sunrise Point-LP PMC (rev 21)
00:1f.3 Audio device: Intel Corporation Sunrise Point-LP HD Audio (rev 21)
00:1f.4 SMBus: Intel Corporation Sunrise Point-LP SMBus (rev 21)
01:00.0 Mass storage controller: Apple Inc. ANS2 NVMe Controller (rev 01)
01:00.1 Non-VGA unclassified device: Apple Inc. T2 Bridge Controller (rev 01)
01:00.2 Non-VGA unclassified device: Apple Inc. T2 Secure Enclave Processor (rev 01)
01:00.3 Multimedia audio controller: Apple Inc. Apple Audio Device (rev 01)
02:00.0 Network controller: Broadcom Inc. and subsidiaries BCM4355 802.11ac Wireless LAN SoC (rev 0c)
03:00.0 PCI bridge: Intel Corporation DSL6540 Thunderbolt 3 Bridge [Alpine Ridge 4C 2015] (rev 06)
04:00.0 PCI bridge: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge 4C 2018] (rev 06)
04:01.0 PCI bridge: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge 4C 2018] (rev 06)
04:02.0 PCI bridge: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge 4C 2018] (rev 06)
04:04.0 PCI bridge: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge 4C 2018] (rev 06)
05:00.0 System peripheral: Intel Corporation JHL7540 Thunderbolt 3 NHI [Titan Ridge 4C 2018] (rev 06)
06:00.0 USB controller: Intel Corporation JHL7540 Thunderbolt 3 USB Controller [Titan Ridge 4C 2018] (rev 06)

AdityaGarg8, 2024-02-22:
```
sudo modprobe -r brcmfmac_wcc
sudo modprobe -r brcmfmac
sleep 5
sudo modprobe brcmfmac
sudo systemctl restart NetworkManager
```

Try this

moshin34, 2024-02-22:
Thank you.  

I entered all the above as instructed, but nothing seems to have changed.   I didn't reboot, which I will now.  But I suspect that will  be moot.

AdityaGarg8, 2024-02-23:
Well since it's an upstream driver, I'd suggest mailing your issue to the Linux kernel maintainers.

If interested, I'll send the email addresses to you.

moshin34, 2024-02-27:
Ok, yes please.   I guess Im just screwed on this model for now?  I should email them and try different hardware?

Thanks!

Redecorating, 2024-02-28:
can you send all the brcmfmac messages between these two:

> brcmf_fw_alloc_request: using brcm/brcmfmac43551c1-pcie for chip BMC4355/12. (in red)

> brcmf_pcie_setup: Dongle setup failed (in red)

rubyonrails3, 2024-03-03:
usbcore: deregistering interface driver brcmfmac
usbcore: registered new interface driver brcmfmac
brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac4364b3-pcie for chip BCM4364/4
brcmfmac 0000:05:00.0: Direct firmware load for brcm/brcmfmac4364b3-pcie.apple,bali-HRPN-u-7.7-X0.bin failed with error -2
Mar 03 18:14:46 MacBookPro kernel: brcmfmac 0000:05:00.0: Direct firmware load for brcm/brcmfmac4364b3-pcie.apple,bali-HRPN-u-7.7.bin failed with error -2
Mar 03 18:14:46 MacBookPro kernel: brcmfmac 0000:05:00.0: Direct firmware load for brcm/brcmfmac4364b3-pcie.apple,bali-HRPN-u.bin failed with error -2
Mar 03 18:14:46 MacBookPro kernel: brcmfmac 0000:05:00.0: Direct firmware load for brcm/brcmfmac4364b3-pcie.apple,bali-HRPN.bin failed with error -2
Mar 03 18:14:46 MacBookPro kernel: brcmfmac 0000:05:00.0: Direct firmware load for brcm/brcmfmac4364b3-pcie.apple,bali-X0.bin failed with error -2
Mar 03 18:14:46 MacBookPro kernel: brcmfmac 0000:05:00.0: Direct firmware load for brcm/brcmfmac4364b3-pcie.apple,bali.bin failed with error -2
Mar 03 18:14:46 MacBookPro kernel: brcmfmac 0000:05:00.0: Direct firmware load for brcm/brcmfmac4364b3-pcie.bin failed with error -2
Mar 03 18:14:46 MacBookPro kernel: brcmfmac 0000:05:00.0: brcmf_pcie_setup: Dongle setup failed
lines 1-23/23 (END)

Redecorating, 2024-03-03:
looking at that, it seems like it can't find the firmware files. maybe check if there's anything in `ls /usr/lib/firmware/brcm | grep apple`.

closed 2024-06-25 by AdityaGarg8