t2linux/T2-Ubuntu #109: Bad substitution error when running iso.sh with sh

dfaw20, 2024-04-07:
Thanks.

When executing the script `iso.sh` using bash (i.e., bash `iso.sh`), it completes successfully. However, running the same script with sh (i.e., `sh iso.sh`) results in a Bad substitution error. This issue seems to stem from sh not supporting certain syntax that is specific to bash, specifically the substring extraction and manipulation within the script. It's important for `iso.sh` to be accessible and user-friendly, even to those not familiar with shell scripting nuances, as users might attempt to run a .sh file with sh and give up if they encounter errors.

### Steps to Reproduce
Create a script named `iso.sh` with the following content:

```
#!/bin/bash
flavour="vanilla"
flavourcap=`echo ${flavour:0:1} | tr '[a-z]' '[A-Z]'`${flavour:1}
echo $flavourcap
```

Make the script executable: `chmod +x iso.sh`
Run the script with bash: `bash iso.sh`
Run the script with sh: `sh iso.sh`

#### Expected Behavior
The script should run successfully with both bash and sh, printing "Vanilla" to the console.

#### Actual Behavior
When run with bash, the script succeeds and prints "Vanilla".
When run with sh, the script fails with the following error message:

```
iso.sh: line 3: Bad substitution
```

### Environment
Operating System: Ubuntu 22.04.4 LTS
Bash version: 5.1.16

### Additional Context
This issue appears to be related to the use of bash-specific features not supported in sh, particularly the use of substring manipulation. For users not deeply familiar with shell scripting, this could lead to confusion and frustration, potentially causing them to give up on using `iso.sh`.

### Proposed Solutions
There are two main approaches to resolving this issue:

**Enhance Documentation**: Clearly specify in the release notes and execution instructions that `iso.sh` should be run with bash (i.e., bash `iso.sh`). This approach provides immediate clarity to the user but does not address the underlying compatibility issue.

**Modify the Script**: Since the issue arises from a non-core logic portion of the script, consider rewriting this part to be compatible with sh. This would involve avoiding bash-specific syntax and features to ensure the script can be executed successfully in both bash and sh environments. This solution enhances the script's accessibility and usability across different shells, making it more beginner-friendly.

AdityaGarg8, 2024-04-07:
If you have ideas, feel free to open a PR.

dfaw20, 2024-04-07:
@AdityaGarg8 
Thanks. I opened a PR #110 .

AdityaGarg8, 2024-04-07:
I've merged the PR. Make sure it runs on macOS as well.

PR #110 (2024-04-07): Make iso.sh Compatible with Both bash and sh
Modified iso.sh to ensure compatibility with both bash and sh. The original script used bash-specific syntax that caused errors in sh. Changes were made to utilize cut and tr commands for string manipulation, avoiding bash-specific features.

## Changes

Used cut to extract the first character of the flavour variable.
Converted the first character to uppercase using tr.
Concatenated the modified first character with the rest of the flavour string using cut.
## Testing

Set the flavour variable (e.g., flavour="Ubuntu").
Execute with bash: bash iso.sh [... cut]
closed 2024-04-07 by AdityaGarg8