t2linux/T2-Ubuntu #124: Auto Startup After Power Failure

webysther, 2024-04-24:
Normally the command works well: `setpci -s 0:1f.0 0xa4.b=0`

Seems not working with Mac mini 2018, have the [v6.5.8-1](https://github.com/t2linux/T2-Ubuntu/releases/tag/v6.5.8-1), anyone knows if works and how works or is not supported yet?

AdityaGarg8, 2024-04-24:
You are using an extremely old kernel. Please update your kernel first

webysther, 2024-04-24:
> You are using an extremely old kernel. Please update your kernel first

I prefer wait for Ubuntu 24 lts tomorrow, any directions from there?

AdityaGarg8, 2024-04-25:
> > You are using an extremely old kernel. Please update your kernel first
> 
> I prefer wait for Ubuntu 24 lts tomorrow, any directions from there?

The kernels shall be released on priority basis. ISOs shall be released after that.

webysther, 2024-04-25:
> > > You are using an extremely old kernel. Please update your kernel first
> > 
> > 
> > I prefer wait for Ubuntu 24 lts tomorrow, any directions from there?
> 
> The kernels shall be released on priority basis. ISOs shall be released after that.

I mean about the auto startup :)

AdityaGarg8, 2024-04-25:
> > > > You are using an extremely old kernel. Please update your kernel first
> > > 
> > > 
> > > I prefer wait for Ubuntu 24 lts tomorrow, any directions from there?
> > 
> > 
> > The kernels shall be released on priority basis. ISOs shall be released after that.
> 
> I mean about the auto startup :)

I cannot say anything about that. In most cases though, such issues get resolved with a kernel update. The latest kernel available is 6.8.7. You must have got the update, provided you set up the apt repo as given in the instructions.

webysther, 2024-04-26:
> I cannot say anything about that. In most cases though, such issues get resolved with a kernel update. The latest kernel available is 6.8.7. You must have got the update, provided you set up the apt [...]

I tried but get error in gcc mismatch...

AdityaGarg8, 2024-04-27:
> I tried but get error in gcc mismatch...

Gcc mismatch? Are you installing Ubuntu noble kernel on an older ubuntu?

webysther, 2024-04-27:
No, jammy kernel, just a dumb gcc 12 error, without the x86_x64 this happens, I will wait for the new image for ubuntu 24 lts and make a clean install.

AdityaGarg8, 2024-04-27:
> No, jammy kernel, just a dumb gcc 12 error, without the x86_x64 this happens, I will wait for the new image for ubuntu 24 lts and make a clean install.

It's released

webysther, 2024-04-28:
I got work after upgrade, thanks!

closed 2024-04-28 by webysther