t2linux/T2-Ubuntu #127: Driver issues on Vega 12 MBP 2019 i9

hlpmenu, 2024-06-23:
Sorry for the lack of exact names and outputs, im on my phone and will update later!

**Short desc**
After the 2nd reboot touchbar stops working due to the gpu not recotnizing it. And huge visual bugs.

**The problem**
On a fresh install, the gpu is working as intended, until the first reboot.

Note that the driver is **not**
changed or replaced.

Im not 100% sure its a driver issue, it could be a gnome, wayland etc issue from some change in noble..

**Symptoms**
- Firstly, theres a discrepancy between how the gpu is showing up in gnome settinga gui, after install it shows up as _"Amd vega 12 [modelnumber] [info]_
While after the first reboot showing up as simply "Amd graphics". But this is only from the gui in gnome settings, but still its a differance.

- If listing screens on first start, it lists 2 screens, the touchbar and the actual screen.

After the first reboot it only lists the actual screen.

- Using tiny-dfr and the "touchbar" command, it firet works, but after a reboot stops working.

The reason being that it panics due to not being able to access the amd gpu du to "device busy".

From what ive seen, its not an issue with the touchbar driver/tiny-dfr or config, but the gpu driver.

- Render/visual bugs

While working fine after install, after the first reboot it breaks.

For some reason it seems to specifically have to do with rendering svgs? Regardless if in browser on a website, in an application etc.

The worst is chromium based browsers, where every single use of a svg anywhere within the browser or websites is completly garbage, looking like the type of glitches you would see on a gpu with dying vram.

While svgs are broken, video playback, images, scaling, is fine.

But i have only chromium, g chrome, firefox and vscode om the machine.

AdityaGarg8, 2024-06-23:
> Sorry for the lack of exact names and outputs, im on my phone and will update later!
>
> **Short desc** After the 2nd reboot touchbar stops working due to the gpu not recotnizing it. And huge visual [...]

Vega 12? Are you using an external GPU?

hlpmenu, 2024-06-23:
> Vega 12? Are you using an external GPU?

Sorry meant radeon pro vega 20!  

Its a:
MBP 2019
i9 9880HK
Radeon Pro Vega 20(not an external vega 12 lol)
1tb
15"

I found that removing all kernel header pkgs, and installing tiny-dfr , then reinstalling the latest t2 kernel fixes the touchbar/display issues at least. 
At least until the next rebbot as far as i know atm.

Messed around a bit with mesa drivers too, which dident help much, but somehow down the line when cleaning up something i did managed to fix the suspend issues that i thought  was just something i had to accept..

But now, when i went back to the macbook, the gpu/touchbar issues were back, and suspend/sleep was once again broken..

But atleast now i have a specific time where it crashed and went back to having issues, so il collect some logs and upload

AdityaGarg8, 2024-06-24:
For touchbar do `sudo touchbar --switch` and restart

hlpmenu, 2024-06-24:
> For touchbar do `sudo touchbar --switch` and restart

No, maybe i was unclear, the issue is on a driver level for the amdgpu..

Initially, the touchbar is seen as a sec screen, and tiny-dfr/touchbar works, same with --switch --restart and so on.
But once the first restart is done, it breaks. 

Ive done 3 setups now, 
**1. Following the guide for linux-t2 ubuntu**
The issue i describe above, high gpu usage when browsing, cant render svgs at all, no touchbar, crash on s2idle.

**2. Purgin all amd gpu drivers and firmware, and using a older mesa driver**
Touchbar working(using tiny-dfr touchbar), suspend working,  but random hard crashes, then a crash where it locked up completly, so not stable.

**3. Just a test case, testing different drivers **
**Using a jammy amdgpu driver + using amd devtools to cap the gpu clockspeed:**
A nightmare to get working, but "worked", except for the fact the gpu did not respect powerlevel, and ran super hot as soon as being used for anything remotly demanding.
Did not crash, but just going to a website with ads made it either overhead until almost freezing the system due to throtteling or simply freezing, until it was moved off screen. 
So basically unusable on a laptop..

**Using amd proprietery drivers**
Worked until i tried vulkan which prevented boot... Dont remember the kernel modules or params and lost the logs..

AdityaGarg8, 2024-06-25:
Did you try linux-t2-lts?

AdityaGarg8, 2024-06-25:
You can also try X11 to rule out whether it's an issue with Wayland

AdityaGarg8, 2024-12-15:
No replies yet, closing. Please open if help still needed.

closed 2024-12-15 by AdityaGarg8