t2linux/T2-Ubuntu #129: USB-C External Display not working properly

decasdev, 2024-07-04:
I've installed Ubuntu 22.04 on my t2 2018 macbook pro 13" with touch bar.
I want to connect my Dell Ultrasharp 25 monitor via USB-C as i always do in macOS, but it is not working properly.
Most of the times it is recognized, but doesn't show an image. And sometimes it randomly works.

AdityaGarg8, 2024-07-04:
Is it thunderbolt?

decasdev, 2024-07-04:
I don't think so.

In regular macOS within system report is shows:

DELL U2520D:
  Resolution:	2560 x 1440 (QHD/WQHD - Wide Quad High Definition)
  UI Looks like:	2560 x 1440 @ 59.00Hz
  Framebuffer Depth:	30-Bit Color (ARGB2101010)
  Display Serial Number:     
  Main Display:	Yes
  Mirror:	Off
  Online:	Yes
  Rotation:	Supported
  Connection Type:	Thunderbolt/DisplayPort

Thunderbolt Bus 0:
   Vendor Name:	Apple Inc.
   Device Name:	MacBook Pro
   UID:
   Route String:	0
   Firmware Version:	47.5
   Domain UUID:
  Port:
   Status:	No device connected
   Link Status:	0x101
   Speed:	Up to 40 Gb/s x1
   Current Link Width:	0x1
   Receptacle:	2
   Link Controller Firmware Version:	1.34.0
 Port:
   Status:	No device connected
   Link Status:	0x101
   Speed:	Up to 40 Gb/s x1
   Current Link Width:	0x1
   Receptacle:	1
   Link Controller Firmware Version:	1.34.0

Thunderbolt Bus 1:
  Vendor Name:	Apple Inc.
  Device Name:	MacBook Pro
  UID:	0x0001CACDA8915301
  Route String:	0
  Firmware Version:	47.5
  Domain UUID:	15D68BD9-FED6-D859-B4FC-81FD664E0DF3
 Port:
   Status:	No device connected
   Link Status:	0x101
   Speed:	Up to 40 Gb/s x1
   Current Link Width:	0x1
   Receptacle:	4
   Link Controller Firmware Version:	1.34.0
 Port:
   Status:	No device connected
   Link Status:	0x101
   Speed:	Up to 40 Gb/s x1
   Current Link Width:	0x1
   Receptacle:	3
   Link Controller Firmware Version:	1.34.0

So i believe it is connected via DisplayPort standard.
In the monitor specs it reads: USB-C (DisplayPort 1.4 mode).

AdityaGarg8, 2024-07-04:
Can you try HDMI. That's what people mostly get working

decasdev, 2024-07-04:
HDMI is not working at all.
I'm using a Satechi USB-C adaptor.

AdityaGarg8, 2024-07-04:
> HDMI is not working at all. I'm using a Satechi USB-C adaptor.

What's `glxinfo | grep "OpenGL"`

AdityaGarg8, 2024-07-04:
And `uname -r`

decasdev, 2024-07-04:
OpenGL vendor string: Intel
OpenGL renderer string: Mesa Intel(R) Iris(R) Plus Graphics 655 (CFL GT3)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 24.1.0-devel (git-a14e611595)
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 4.6 (Compatibility Profile) Mesa 24.1.0-devel (git-a14e611595)
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 24.1.0-devel (git-a14e611595)
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:

6.9.7-4-t2-jammy

AdityaGarg8, 2024-07-04:
Are you using iGPU?

decasdev, 2024-07-04:
Yes, it is my only GPU

AdityaGarg8, 2024-07-04:
Try this

```
sudo apt update
sudo apt purge linux-t2 linux-headers-$(uname -r) linux-image-$(uname -r)
sudo apt install linux-t2=6.9.7-1-jammy
```

Then restart and try HDMI

Also share `uname -r` and `cat /proc/cmdline` after restart.

decasdev, 2024-07-04:
The HDMI is working now.
Thank you!

uname -r
6.9.7-4-t2-jammy

cat /proc/cmdline
BOOT_IMAGE=/boot/vmlinuz-6.9.7-4-t2-jammy root=UUID=ab2ccffe-9160-4113-a066-b7979c4f0fba ro pcie_ports=native intel_iommu=on iommu=pt quiet splash i915.enable_hangcheck=0 i915.request_timeout_ms=200000 vt.handoff=7

I would love to get the USB-c connection working properly tho. Do you have any idea why this might be happening?
It detects the monitor, it just doesn't output an image:
xrandr --listmonitors
Monitors: 2
 0: +*XWAYLAND8 2560/550x1440/310+0+0  XWAYLAND8
 1: +XWAYLAND0 2560/290x1600/180+2560+0  XWAYLAND0

Well, sometimes it does, sometimes it doesn't.

AdityaGarg8, 2024-07-04:
> The HDMI is working now. Thank you!
>
> uname -r 6.9.7-4-t2-jammy
>
> cat /proc/cmdline BOOT_IMAGE=/boot/vmlinuz-6.9.7-4-t2-jammy root=UUID=ab2ccffe-9160-4113-a066-b7979c4f0fba ro pcie_ports=native [...]

File a bug report in upstream kernel for that

> Do you have any idea why this might be happening? It detects the monitor, it just doesn't output an image: xrandr --listmonitors Monitors: 2 0: +*XWAYLAND8 2560/550x1440/310+0+0 XWAYLAND8 1: +XWAYLAND0 2560/290x1600/180+2560+0 XWAYLAND0
> 
> Well, sometimes it does, sometimes it doesn't.

closed 2024-07-04 by AdityaGarg8