t2linux/T2-Ubuntu #131: MacBookPro16,2 appletbdrm does not bind to touchbar display

chithanh, 2024-08-02:
When booting the T2 Linux Ubuntu image (24.04, kernel 6.9.5-1-t2-noble) on a MacBook Pro 16,2 the touchbar stays dark, doesn't accept input, and `appletbdrm` is not loaded. Loading the `appletbdrm` module manually makes no difference. It appears that the driver either does not include the correct hardware ID for the touch bar, or that the device is somehow hidden from the USB subsystem.

[dmesg](https://github.com/user-attachments/files/16469966/dmesg.txt)
[lspci -nnk](https://github.com/user-attachments/files/16469967/lspci-k.txt)
[lsusb -v](https://github.com/user-attachments/files/16469968/lsusb.txt)
[lsusb -v -t](https://github.com/user-attachments/files/16469969/lsusb-t.txt)
[modinfo appletbdrm](https://github.com/user-attachments/files/16469970/modinfo_appletbdrm.txt)

AdityaGarg8, 2024-08-02:
> When booting the T2 Linux Ubuntu image (24.04, kernel 6.9.5-1-t2-noble) on a MacBook Pro 16,2 the touchbar stays dark, doesn't accept input, and `appletbdrm` is not loaded. Loading the `appletbdrm` [...]

Did you install `tiny-dfr`?

chithanh, 2024-08-02:
Yes, `tiny-dfr-0.3.0-1-noble` is installed, but it does not find the dri node, presumably because `appletbdrm` didn't create it (`card0` is the Intel iGPU).

```
$ sudo tiny-dfr 
thread 'main' panicked at src/main.rs:518:43:
called `Result::unwrap()` on an `Err` value: No touchbar device found, attempted: [
    /dev/dri/card0: Device or resource busy (os error 16)
]
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
$ ls -l /dev/dri/
total 0
drwxr-xr-x  2 root root         80 Aug  2 14:32 by-path
crw-rw----+ 1 root video  226,   0 Aug  2 14:50 card0
crw-rw----+ 1 root render 226, 128 Aug  2 14:50 renderD128
```

AdityaGarg8, 2024-08-02:
> Yes, `tiny-dfr-0.3.0-1-noble` is installed, but it does not find the dri node, presumably because `appletbdrm` didn't create it (`card0` is the Intel iGPU).
>
> ```
> $ sudo tiny-dfr
> thread 'main' [...]

If you run `sudo touchbar --switch` and restart you Mac, does it still work?

AdityaGarg8, 2024-08-02:
Btw, if you are running `sudo tiny-dfr`, you have to first disable the systemd service (`sudo systemctl disable --now tiny-dfr`)

AdityaGarg8, 2024-08-02:
> Btw, if you are running `sudo tiny-dfr`, you have to first disable the systemd service (`sudo systemctl disable --now tiny-dfr`), else it will show the error.

chithanh, 2024-08-02:
The systemd `tiny-dfr.service` fails to start anyway and `tiny-dfr` is not running according to `ps aux`. I am trying `touchbar --switch` next.

chithanh, 2024-08-02:
As I am booting the LiveUSB image, it didn't make a difference. With some tinkering (blacklisting `apple_bce` at boot, then loading `apple_ibridge`, `apple_touchbar`, and `apple_bce` in that order, I got the following output:

```
[  343.119515] usb 5-6: New USB device found, idVendor=05ac, idProduct=8302, bcdDevice= 2.01
[  343.119521] usb 5-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  343.119523] usb 5-6: Product: Touch Bar Display
[  343.119524] usb 5-6: Manufacturer: Apple Inc.
[  343.119525] usb 5-6: SerialNumber: 0000000000000000
[  343.126671] input: Apple Inc. Touch Bar Display as /devices/pci0000:00/0000:00:1c.4/0000:e6:00.1/apple-bce/apple-bce/bce-vhci/usb5/5-6/5-6:1.0/0003:05AC:8302.000A/input/input15
[  343.127695] bce-vhci: [00] URB failed: 3
[  343.186492] apple-touchbar 0003:05AC:8302.000A: input: USB HID v1.01 Keyboard [Apple Inc. Touch Bar Display] on usb-bce-vhci-6/input0
[  343.186886] bce_vhci_reset_device 7
[  343.318367] usb 5-7: new high-speed USB device number 8 using apple-bce
[  343.318372] bce_vhci_enable_device
[  343.318859] bce_vhci_cmd_device_create 7 -> 7
[  343.321240] bce_vhci_reset_device 7
[  343.321242] bce-vhci: Not implemented: wait for pending output requests
[  343.450478] bce_vhci_enable_device
[  343.478207] usb 5-7: New USB device found, idVendor=05ac, idProduct=8102, bcdDevice= 2.01
[  343.478219] usb 5-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  343.478223] usb 5-7: Product: Touch Bar Backlight
[  343.478227] usb 5-7: Manufacturer: Apple Inc.
[  343.478230] usb 5-7: SerialNumber: 0000000000000000
[  343.487572] apple-touchbar 0003:05AC:8102.000B: : USB HID v1.01 Device [Apple Inc. Touch Bar Backlight] on usb-bce-vhci-7/input0
[  343.569235] hid-apple-magic-backlight 0003:05AC:8102.000C: hiddev6,hidraw9: USB HID v1.01 Device [Apple Inc. Touch Bar Backlight] on usb-bce-vhci-7/input1
[  348.742551] apple-touchbar 0003:05AC:8302.000A: tb: Failed to set touch bar mode to 2 (-110)
```

If necessary, I can see if I can locate another USB drive to make a proper install to. But if I am not mistaken, `appletbdrm` still needs to bind to the hardware and create a `/dev/dri` node before the touchbar can work.

AdityaGarg8, 2024-08-02:
> As I am booting the LiveUSB image, it didn't make a difference. With some tinkering (blacklisting `apple_bce` at boot, then loading `apple_ibridge`, `apple_touchbar`, and `apple_bce` in that order), the touchbar still does not work and I got the following output:
> 
> ```
> [  343.119515] usb 5-6: New USB device found, idVendor=05ac, idProduct=8302, bcdDevice= 2.01
> [  343.119521] usb 5-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [  343.119523] usb 5-6: Product: Touch Bar Display
> [  343.119524] usb 5-6: Manufacturer: Apple Inc.
> [  343.119525] usb 5-6: SerialNumber: 0000000000000000
> [  343.126671] input: Apple Inc. Touch Bar Display as /devices/pci0000:00/0000:00:1c.4/0000:e6:00.1/apple-bce/apple-bce/bce-vhci/usb5/5-6/5-6:1.0/0003:05AC:8302.000A/input/input15
> [  343.127695] bce-vhci: [00] URB failed: 3
> [  343.186492] apple-touchbar 0003:05AC:8302.000A: input: USB HID v1.01 Keyboard [Apple Inc. Touch Bar Display] on usb-bce-vhci-6/input0
> [  343.186886] bce_vhci_reset_device 7
> [  343.318367] usb 5-7: new high-speed USB device number 8 using apple-bce
> [  343.318372] bce_vhci_enable_device
> [  343.318859] bce_vhci_cmd_device_create 7 -> 7
> [  343.321240] bce_vhci_reset_device 7
> [  343.321242] bce-vhci: Not implemented: wait for pending output requests
> [  343.450478] bce_vhci_enable_device
> [  343.478207] usb 5-7: New USB device found, idVendor=05ac, idProduct=8102, bcdDevice= 2.01
> [  343.478219] usb 5-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [  343.478223] usb 5-7: Product: Touch Bar Backlight
> [  343.478227] usb 5-7: Manufacturer: Apple Inc.
> [  343.478230] usb 5-7: SerialNumber: 0000000000000000
> [  343.487572] apple-touchbar 0003:05AC:8102.000B: : USB HID v1.01 Device [Apple Inc. Touch Bar Backlight] on usb-bce-vhci-7/input0
> [  343.569235] hid-apple-magic-backlight 0003:05AC:8102.000C: hiddev6,hidraw9: USB HID v1.01 Device [Apple Inc. Touch Bar Backlight] on usb-bce-vhci-7/input1
> [  348.742551] apple-touchbar 0003:05AC:8302.000A: tb: Failed to set touch bar mode to 2 (-110)
> ```
> 
> If necessary, I can see if I can locate another USB drive to make a proper install to. But if I am not mistaken, `appletbdrm` still needs to bind to the hardware and create a `/dev/dri` node before the touchbar can work.

Late loading of apple-bce is an extremely bad idea. This module should instead be loaded as soon as possible.

And didn't you do `touchbar --switch` in a proper installation?

Why do you have `hid-apple-magic-backlight`, this driver is no longer used. What ISO are you using? What drivers are you trying to install?

Talking about `appletbdrm`, the driver is written perfectly. Touchbar on MacBooks have 2 modes: 1. Windows mode with the basic media controls and F keys and 2. The fancy macOS mode which makes it a second display.

By default the touchbar enters the Windows mode and is managed by `hid-appletb-kbd` driver. Basically the driver is quite stripped off so Windows mode doesn't really exist now. To make it work in the macOS mode, we use a udev rule to bind it to the `appletbdrm` driver and it is shipped with tiny-dfr.

I believe you are using some old iso, or there is something else wrong here.

AdityaGarg8, 2024-08-02:
`touchbar --switch` blacklists all the touchbar drivers and installs another driver named `apple-touchbar`. This is a fully fledged working driver for the Windows mode. If you are not getting this to work, there is something seriously wrong with your setup. Maybe you messed up something? Maybe you are doing something different from the standard ways?

And DO NOT message with `apple-bce`. It's function is to expose your inbuilt USB devices which includes your touchbar, trackpad and keyboard.

AdityaGarg8, 2024-08-02:
I would appreciate the answers to these questions:

1. What ISO are you using?
2. Before installing tiny-dfr, did you try to follow some random guide to get the touchbar working?
3. After installing tiny-dfr, did you restart?
4. You mentioned live iso in an answer before. Do you know that you need a proper reboot which is possible in a persistent storage to make it effective?

AdityaGarg8, 2024-08-02:
It would also be nice if you share your complete `journalctl`

chithanh, 2024-08-02:
> Talking about `appletbdrm`, the driver is written perfectly.

But `modinfo appletbdrm` says `alias:          usb:v05ACp8302d*dc*dsc*dp*ic10isc*ip*in*` which I understand means it is supposed to be loaded when such hardware is present and bind to it. But that doesn't happen.

> What ISO are you using?

I'm using `ubuntu-24.04-6.9.5-t2-noble.iso` from https://github.com/t2linux/T2-Ubuntu/releases which I realize is not the latest, but fairly recent. I will do all further tests with the very latest ISO.

> Before installing tiny-dfr, did you try to follow some random guide to get the touchbar working?

No. I only ran `modprobe appletbdrm` then `apt update` then `apt install tiny-dfr`.

> After installing tiny-dfr, did you restart?

> You mentioned live iso in an answer before. Do you know that you need a proper reboot which is possible in a persistent storage to make it effective?

With that ISO, there seems to be no persistent storage besides the system logs.

> From where did hid-apple-magic-backlight come from?

I did not install it explicitly.

chithanh, 2024-08-02:
Ok, when blacklisting the correct drivers then the touchbar works with `apple-touchbar`, without a reboot.

`journalctl` output after `sudo apt update && sudo apt install tiny-dfr`

[journalctl](https://github.com/user-attachments/files/16475416/journalctl.txt) (touchbar not working)

After booting with `modprobe.blacklist=appletbdrm,hid_appletb_kbd,hid_appletb_bl`
then `sudo apt update && sudo apt install tiny-dfr`
then `sudo touchbar --switch`
then `sudo modprobe apple-ibridge && sudo modprobe apple-touchbar` (`apple-ibridge` might be unnecessary)
then the touchbar works (in basic mode).

[journalctl](https://github.com/user-attachments/files/16475608/journalctl2.txt) (touchbar working with `apple-touchbar`)

AdityaGarg8, 2024-08-03:
> I found the option for persistence, I had to add `persistent` to kernel boot parameters, by default this was off.
> 
> With persistence and after reboot, the `appletbdrm` driver now binds to interface 1 of USB device `05ac:8302` which it didn't do before, and the touchbar works.
> 
> If I blacklist `appletbdrm` and load it manually after boot, then it will not work. Maybe a race condition?

Yes it's probably a race condition here. If you don't want persistent storage, you'll have to modify the iso to have tiny-dfr by default, so that the udev rule, which works only at boot becomes effective.

> 
> In any case, I guess the issue can be closed now. There is probably no problem in `appletbdrm` itself, as you wrote correctly in [#131 (comment)](https://github.com/t2linux/T2-Ubuntu/issues/131#issuecomment-2265833845)

closed 2024-08-03 by AdityaGarg8