t2linux/T2-Ubuntu #132: iso.sh says "Please connect to the internet" when already connected

katarafett, 2024-08-28:
I'm using Arch Linux, and I'm connected to the internet via ethernet.

On running `iso.sh`, I get the message `Please connect to the internet`. After commenting out lines 19 through 24:

```bash
# if [ $? -eq 0 ]; then
#     true
# else
#     echo "Please connect to the internet"
#     exit 1
# fi
```

it ran just fine.

AdityaGarg8, 2024-08-29:
Script needs netcat to test internet access. You probably don't have it installed.

https://man.archlinux.org/man/nc.1.en

lemmyg, 2024-11-06:
Hello @AdityaGarg8 ,
I got this project if you want to give a try.
you need to download kubuntu official iso from the web site.
https://github.com/lemmyg/t2-kubuntu/tree/test

AdityaGarg8, 2024-11-06:
> Hello @AdityaGarg8 , I got this project if you want to give a try. you need to download kubuntu-24.10-desktop-amd64.iso from the web site. https://github.com/lemmyg/t2-kubuntu/tree/test

Does the installer work as well?

lemmyg, 2024-11-06:
Yes

lemmyg, 2024-11-06:
I have tested the installation in virtual box and the live session in my computer. I didn't install in my computer yet. But it should work.

AdityaGarg8, 2024-11-07:
> Hello @AdityaGarg8 , I got this project if you want to give a try. you need to download kubuntu-24.10-desktop-amd64.iso from the web site. https://github.com/lemmyg/t2-kubuntu/tree/test

Can you give me instructions please?

lemmyg, 2024-11-07:
yes,
you need to download kubuntu 24.10 iso from https://kubuntu.org/getkubuntu/

git clone -b test https://github.com/lemmyg/t2-kubuntu.git
cd t2-kubuntu/
edit ISO_IMAGE with the path to your iso file
sh build_in_docker.sh
note that you may need to run twice as at the first time fails to mount the iso. I dont know why.

it will create the output folder with kubuntu-24.10-t2-desktop-amd64.iso file
sudo chown -R $USER:$USER output
test kubuntu-24.10-t2-desktop-amd64.iso in virtual box for example

AdityaGarg8, 2024-11-07:
===]> Info: Generating final ISO ... 
grub-mkrescue: error: `mformat` invocation failed

workflow: https://github.com/AdityaGarg8/random/actions/runs/11725595328/job/32662273856#step:4:673

lemmyg, 2024-11-07:
ah sorry I put my comment in the wrong issue
lets continue in the original one