t2linux/T2-Ubuntu #133: Endeavour OS on Macbook Air 2020

wmrose, 2024-09-23:
Wifi will not connect.  Journalctl shows "NetworkManager [559]: <warn> [1727127846.1600] platform-linux: do-change-link[2]: failure 12 (cannot allocate memory)
and
...device (wlan0): sethw-addr: failed to set MAC address...

Drivers seem to be where they belong, in /lib/firmware/brcm

Different computer or I'd paste output.

Thanks, Bill

AdityaGarg8, 2024-09-24:
This is an issue with Endeavour OS, so please open it in their repo. This repo is for Ubuntu.

AdityaGarg8, 2024-09-24:
https://github.com/t2linux/EndeavourOS-ISO-t2

closed 2024-09-24 by AdityaGarg8

wmrose, 2024-09-24:
Sorry!  For what it's worth, although NMCLI doesn't work, it turns out that IWC does.  Could be helpful with your distro, too.  Found on Prohoster.info's blog.

NoaHimesaka1873, 2024-09-24:
@wmrose That's a pretty well-known issue, affecting *all* Broadcom devices, caused by newest version of wpa_supplicant.

AdityaGarg8, 2024-09-26:
> Sorry! For what it's worth, although NMCLI doesn't work, it turns out that IWC does. Could be helpful with your distro, too. Found on Prohoster.info's blog.

I believe you are facing the infamous wpa_supplicant regression issue. Fortunately Ubuntu isn't affected yet since it uses the older version of wpa_supplicant