t2linux/T2-Ubuntu #134: No ethernet in kubuntu isos

lemmyg, 2024-10-27:
Hello @AdityaGarg8 ,
I was trying reinstall kubuntu 24.04 on my laptop. I found that for some reason in the KDE live session or the installed one, the ethernet device was missing. So I have no access to internet to download pkgs etc. Could it be some missing KDE packages although everything seems correct.
I was wondering if we should install KDE desktop with the  --install-recommends option as some packages were moved from required to recommended.

This issue doesn't not happen in Ubuntu isos and it can be reproduced in the virtual box, by loading kubuntu live session.
Thanks,

Lemmyg

AdityaGarg8, 2024-10-27:
Hi @lemmyg 

I don't really know much about kde isos tbh, but feel free to make a PR and I'll be happy to merge it.

lemmyg, 2024-10-27:
I'll do some tests this week.
Thanks

AdityaGarg8, 2024-10-27:
You also would like to check the commit history of mainline branch for packages, since the kde 24.04 isos are now being made from the LTS branch and I had to manually change jammy to noble again in that branch.

lemmyg, 2024-10-27:
Increase the size of the iso is not a problem any more, right? As you are splitting automatically in multiple files. Due the --install-recommends

AdityaGarg8, 2024-10-28:
Not an issue

lemmyg, 2024-10-28:
Hello I did some tests today.
add --install-recommends and restore iwd that was remove since my merge for Kubuntu:
Also remove grub kernel options.
I have checked kde network pkgs. they are installed.
no luck. network device is there but not used.
the networkmanger config seems correct. not managed.
I have a laptop with those settings and seems ok.
mistery! :)

lemmyg, 2024-10-31:
Hello,
I did some findings. Basically the problem is that network manager doesnt create the ethernet connection automatically.
to solve this, user need to create one from Kde and then restart manually network manager service.
I have check network manager config and I dont see any diference from normal installation. All seems correct.
Another problem I found is that Kde installation is not completed and many gnome libraries are there because the iso is generated based on ubuntu docker.

I did a test install kubuntu directly from the official iso and then add the missing packages. 
Im wondering if would not better create an iso from the official one and add t2 kernel pkgs.

AdityaGarg8, 2024-10-31:
> Hello, I did some findings. Basically the problem is that network manager doesnt create the ethernet connection automatically. to solve this, user need to create one from Kde and then restart manually network manager service. I have check network manager config and I dont see any diference from normal installation. All seems correct. Another problem I found is that Kde installation is not completed and many gnome libraries are there because the iso is generated based on ubuntu docker.
> 
> I did a test install kubuntu directly from the official iso and then add the missing packages. Im wondering if would not better create an iso from the official one and add t2 kernel pkgs.
> 
> By the way, your script get firmware in ubuntu works perfectly. :)

It's possible to create from official iso, but I haven't been able to get the installer working like that.

Is there any terminal command that can fix do the Ethernet fix you did?

lemmyg, 2024-10-31:
# to find the device
nmcli device status
#create the connection
sudo nmcli connection add type ethernet ifname <device> con-name MyEthernet
#change the proority to mimic kde settings
sudo nmcli connection modify MyEthernet connection.autoconnect-priority -100

lemmyg, 2024-10-31:
about creating an iso. i got some steps from chatgpt. ill do some tests

lemmyg, 2024-11-04:
Hello @AdityaGarg8 ,
I have managed to create iso based in the official kubuntu and add all your change. Tested in Virtual box and live session, seems to work. Now trying convert the bash script to a docker script to avoid chroot the local computer.
I'll let you know when is ready.

AdityaGarg8, 2024-11-07:
===]> Info: Generating final ISO ...
grub-mkrescue: error: mformat invocation failed

workflow: https://github.com/AdityaGarg8/random/actions/runs/11725595328/job/32662273856#step:4:673

lemmyg, 2024-11-07:
try with kubuntu-24.10 for now.
Isos could be different between versions.
For now, I prefer just to test that we can modify an official iso.

AdityaGarg8, 2024-11-07:
> try with kubuntu-24.10 for now. Isos could be different between versions. For now, I prefer just to test that we can modify an official iso.

Same error

https://github.com/AdityaGarg8/random/actions/runs/11727657123/job/32669257079

AdityaGarg8, 2024-11-07:
You may want to change the Ubuntu version to 24.10 in `build_in_docker.sh`

AdityaGarg8, 2024-11-07:
Managed to make an ISO, https://nightly.link/AdityaGarg8/random/actions/runs/11728381773/Kubuntu.zip

Will test this when I get free.

AdityaGarg8, 2024-11-07:
> Great thanks. Ideally the iso name should be defined by the official iso. I did new test could boot in virtual box. However for some reason with the iso generated using docker. I can't make a bootable usb. This was working before. Probably is relate to some missing change on boot files.

The instructions you gave me also needed docker, so I dunno what's wrong :(

lemmyg, 2024-11-07:
It should be something else.
I'll do some tests on the weekend.

-------- Original Message --------

lemmyg, 2024-11-13:
hey @AdityaGarg8 ,
finally I got a working version
https://github.com/lemmyg/t2-kubuntu/actions/runs/11822961515

I have burn in usb and boot with my mac

AdityaGarg8, 2024-11-13:
> hey @AdityaGarg8 , finally I got a working version https://github.com/lemmyg/t2-kubuntu/actions/runs/11822961515
> 
> I have burn in usb and boot with my mac

You want to maintain the Kubuntu isos seperately or merge it to t2-ubuntu?

AdityaGarg8, 2024-11-13:
> hey @AdityaGarg8 , finally I got a working version https://github.com/lemmyg/t2-kubuntu/actions/runs/11822961515
> 
> I have burn in usb and boot with my mac

Btw I'll test it when I get time 🙂

lemmyg, 2024-11-22:
Hello @AdityaGarg8 ,
Finally I tested the iso in my computer. All good. Also I managed to install it in a Luks partition.

This is good to merge it in your repo. Let me know how you would like to proceed.

Thanks
Galder

AdityaGarg8, 2024-11-22:
> Hello @AdityaGarg8 , Finally I tested the iso in my computer. All good. Also I managed to install it in a Luks partition.
> 
> This is good to merge it in your repo. Let me know how you would like to proceed.
> 
> Thanks Galder

Hi

Thanks for testing
Sorry I couldn't test the iso since my exams are near. I believe you have write access to the T2-Ubuntu repo. You can modify the kubuntu branches there yourself. Just make sure that the CI splits the ISO into 2 parts and releases them similarly as now.

lemmyg, 2024-11-22:
I'll give a try.

Thanks

AdityaGarg8, 2024-11-26:
Hi @lemmyg 

Can you test the iso here please:

https://github.com/AdityaGarg8/t2-kubuntu/actions/runs/12034772710/job/33552130954

AdityaGarg8, 2024-11-26:
Also looks like Ubuntu ISOs are kinda different, so your method for Kubuntu doesn't seem to work on Ubuntu

lemmyg, 2024-11-26:
Should not be much different. 
I'll have a look.

lemmyg, 2024-11-26:
whats the issue?

lemmyg, 2024-11-26:
Unfortunately, Ubuntu iso is way more complicated. The file system is divided in multiple squashfs files.
I don't think I will be able to build it. Too tricky.

AdityaGarg8, 2024-11-27:
https://github.com/AdityaGarg8/t2-kubuntu/releases/tag/v6.11.7-1

Finally made a release. Testing appreciated. Will merge to the main repo then.

lemmyg, 2024-11-27:
I see you removed lines:
 
 # Add udev Rule for AMD GPU Power Management
cat <<EOL > /etc/udev/rules.d/30-amdgpu-pm.rules
KERNEL=="card[012]", SUBSYSTEM=="drm", DRIVERS=="amdgpu", ATTR{device/power_dpm_force_performance_level}="low"
EOL

# Add Required Modules to initramfs for Apple Keyboard
cat <<EOL >> /etc/initramfs-tools/modules
# Required modules for getting the built-in apple keyboard to work:
snd
snd_pcm
apple-bce
EOL
update-initramfs -u

# Blacklist Modules for T2 Chip Internal Ethernet
cat <<EOL >> /etc/modprobe.d/blacklist.conf
# Disable for now T2 chip internal USB ethernet
blacklist cdc_ncm
blacklist cdc_mbim
EOL

those lines are required for me. especially the gpu. without that my gpu will fail at booting.

the rest are just tweaks that usually need to be done after the installation.
So having them will safe users time

AdityaGarg8, 2024-11-27:
> I see you removed lines:
> 
> # Add udev Rule for AMD GPU Power Management
> cat < /etc/udev/rules.d/30-amdgpu-pm.rules KERNEL=="card[012]", SUBSYSTEM=="drm", DRIVERS=="amdgpu", ATTR{device/power_dpm_force_performance_level}="low" EOL
> 

Is the failed boot because of the external monitor or it occurs every time?

> # Add Required Modules to initramfs for Apple Keyboard
> cat <> /etc/initramfs-tools/modules
> 
> # Required modules for getting the built-in apple keyboard to work:
> snd snd_pcm apple-bce EOL update-initramfs -u

This is needed for encryption only. Else not needed. Also, this setting breaks touchbar for many, so make it optional
> 
> # Blacklist Modules for T2 Chip Internal Ethernet
> cat <> /etc/modprobe.d/blacklist.conf
> 
> # Disable for now T2 chip internal USB ethernet
> blacklist cdc_ncm blacklist cdc_mbim EOL

This blacklist workaround is no longer needed, instead we have a udev rule now, which I'll add later.

> 
> those lines are required for me. especially the gpu. without that my gpu will fail at booting.
> 
> the rest are just tweaks that usually need to be done after the installation. So having them will safe users time

lemmyg, 2024-11-27:
I did a quick test with https://github.com/AdityaGarg8/t2-kubuntu/actions/runs/12034772710

I can boot in virtual box with efi boot loader

About the gpu rule. My gpu requires to be at low profile other ways I get a black screen at booting or shutdown.
Not related to the external monitor.

AdityaGarg8, 2024-11-27:
> I did a quick test with https://github.com/AdityaGarg8/t2-kubuntu/actions/runs/12034772710
> 
> I can boot in virtual box with efi boot loader
> 
> About the gpu rule. My gpu requires to be at low profile other ways I get a black screen at booting or shutdown. Not related to the external monitor.

What GPU are you using?

lemmyg, 2024-11-27:
03:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Navi 14 [Radeon RX 5500/5500M / Pro 5500M] (rev 40)

lemmyg, 2024-11-27:
I never used that one

AdityaGarg8, 2024-11-27:
I cannot add the udev rule in the iso, since it's meant to be used by all

AdityaGarg8, 2024-11-27:
> that rule should not cause any issue. It might even help with the heating issues that many users have. Also a note could be added if anyone would like to remove it.

Yeah, but it makes the GPU use less power, I.e. less performance. Majority wants GPU intensive tasks, not low performance.

AdityaGarg8, 2024-11-27:
You can always use the safe graphics option to be able to install

lemmyg, 2024-11-27:
I use it and I dont notice anything. it works just fine. maybe less heating.

AdityaGarg8, 2024-11-27:
> I use it and I dont notice anything. it works just fine. maybe less heating.

You are the only one having issue without this, so this workaround is specifically for you. Similarly, there are some people having issues, say for wifi not working on boot, we have workarounds for them. So the ISO cannot keep everyone happy, but atleast I don't want to force workarounds on all.

Adding something due to some issue is easier than removing something not needed.

lemmyg, 2024-11-27:
ok fine. Ill use own variant.
leave it commented

AdityaGarg8, 2024-11-27:
Btw can you disable the splash screen and share the logs so that I can find out what exactly is going on with the driver?

AdityaGarg8, 2024-11-27:
> I got a kernel panic.

Hmm, `journalctl -k -b -1` should give the logs of the previous boot. Please share that.

lemmyg, 2024-11-27:
no, its fine. I dont want to waste time with this.

just add a comment in the line.

lemmyg, 2024-11-27:
I guess we are good to close this issue.

thanks for you help @AdityaGarg8

AdityaGarg8, 2024-11-27:
I'll take some feedbacks from the discord regarding power management. If people support it, I'll add it. So keep it open till then

AdityaGarg8, 2024-12-15:
New ISOs, finally released. Thanks for the help @lemmyg .

closed 2024-12-15 by AdityaGarg8

lemmyg, 2024-12-15:
Thanks for helping with the CI @AdityaGarg8