t2linux/T2-Ubuntu #136: Black screen after the fresh install Ubuntu 24.04, the latest v6.12.4-1

Rom888, 2024-12-14:
I decided to do a fresh upgrade from Ubuntu 23.10 to Ubuntu 24.04.

The first issue:
When I selected "install Ubuntu" an error message appeared like:
'Ubiquity service failed to start'.
Ubuntu loaded, and then I installed it using the install app.

The second issue:
After rebooting, Ubuntu showed its logo and after that I only see a black screen.

machine:
```
macOs: Ventura 13.2

Ubuntu 24.04
v6.12.4-1

MacBookPro16.1
Intel Core i9 2.3g

hybrid graphics:
Intel UHD graphics 630
AMD radeon pro 5500M
```

lemmyg, 2024-12-14:
Hello,
same issue happens to me as I have same MacBook and GPU model.
The workaround for this is boot in recovery mode and add following rule.

nano /etc/udev/rules.d/30-amdgpu-pm.rules

KERNEL=="card[012]", SUBSYSTEM=="drm", DRIVERS=="amdgpu", ATTR{device/power_dpm_force_performance_level}="low"

You could also mount the partion with live usb and add the rule

AdityaGarg8, 2024-12-15:
@Rom888 , has the workaround suggested by @lemmyg helped?

Rom888, 2024-12-15:
@AdityaGarg8 , yes, the workaround works.
It would be really useful to have this rule included in the releases.

AdityaGarg8, 2024-12-15:
I'll add it to the next release after testing on my end.

AdityaGarg8, 2024-12-20:
Added to latest release

closed 2024-12-20 by AdityaGarg8