t2linux/T2-Ubuntu #140: No wifi after installation

Jynra, 2024-12-24:
Hello,
Even after installation I can't connect the wifi. There is only the wired option.
And I don't have the touchbar either.
Is there an additional installation to do ?

AdityaGarg8, 2024-12-24:
Did you run the firmware script for wifi?

For touchbar, did you install tiny-dfr or are using basic mode?

Jynra, 2024-12-25:
> Did you run the firmware script for wifi?
> 
> For touchbar, did you install tiny-dfr or are using basic mode?

During installation I selected “install third-party software for graphics and wi-fi hardware”.
Is there an extra step after reboot ?
What is and where is the firmware script for wifi ?
I didn't install tiny-dfr, it's not indicated anywhere in the T2linux procedure.

AdityaGarg8, 2024-12-25:
> During installation I selected “install third-party software for graphics and wi-fi hardware”. Is there an extra step after reboot ? What is and where is the firmware script for wifi ? I didn't [...]

https://wiki.t2linux.org/distributions/ubuntu/installation/#setting-up-wi-fi-and-bluetooth

I believe you didn't follow this guide.

AdityaGarg8, 2024-12-25:
For touchbar, share `lsusb`

Jynra, 2024-12-27:
Thank you
I didn't read the right guide.
probleme solved for the wifi and touchbar.

closed 2024-12-27 by AdityaGarg8