t2linux/T2-Ubuntu #141: Closing the screen causes a black screen on reopening

Jynra, 2024-12-27:
When I close my screen and open it again, nothing appears and the computer restarts.
Is there a solution to this problem ?

AdityaGarg8, 2024-12-27:
> When I close my screen and open it again, nothing appears and the computer restarts. Is there a solution to this problem ?

Suspend is broken on T2 Macs, no solution.

Please read https://wiki.t2linux.org/state/

You may change the lid behaviour to shut down, lock etc. Google how to do that on Ubuntu.

Jynra, 2024-12-27:
Ok, thks
For other people :
```sudoedit /etc/systemd/logind.conf```
Find the line ```#HandleLidSwitch```
And choice your preference :
```
    HandleLidSwitch=lock (lock when lid closed)

    HandleLidSwitch=ignore (do nothing)

    HandleLidSwitch=poweroff (shutdown)

    HandleLidSwitch=hibernate (hibernate Ubuntu)
```

closed 2024-12-27 by AdityaGarg8