t2linux/T2-Ubuntu #143: Ubuntu 24.04 and 24.10 stuck on boot most of the times on iMac 2020 with latest Sequoia (15.2)

bigfoot7375, 2024-12-31:
HW: iMac 2020 27" (Intel i5) / AMD Radeon Pro 5300 / 128GB RAM
SW: Tried both BigSur 11.7.10 and Sequoia 15.2
Issue: Both installer and Installed OS expereince same system hang
Details: iMac boots successfully once in few boot attempts.
after being stubborn, successfuly loaded installer and installed Ubuntu.
Installed OS behave as installer - one successful boot after few boot attempts (need long press on power button to reset).
Not sure if helpful, but attaching terminal screenshot appearing while boot fail.
[image]

AdityaGarg8, 2024-12-31:
> HW: iMac 2020 27" (Intel i5) / AMD Radeon Pro 5300 / 128GB RAM SW: Tried both BigSur 11.7.10 and Sequoia 15.2 Issue: Both installer and Installed OS expereince same system hang Details: iMac boots successfully once in few boot attempts. after being stubborn, successfuly loaded installer and installed Ubuntu. Installed OS behave as installer - one successful boot after few boot attempts (need long press on power button to reset). Not sure if helpful, but attaching terminal screenshot appearing while boot fail. ![IMG_0103](https://private-user-images.githubusercontent.com/75070545/399461823-75898cfd-9d16-4a38-8b3d-e1944f6491ac.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU2NDQ3MTAsIm5iZiI6MTczNTY0NDQxMCwicGF0aCI6Ii83NTA3MDU0NS8zOTk0NjE4MjMtNzU4OThjZmQtOWQxNi00YTM4LThiM2QtZTE5NDRmNjQ5MWFjLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjMxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIzMVQxMTI2NTBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wYzk4YmY2OTliZjA5N2VhYzViYjhiYzkyMzQwYzQyOTBiMjI1Yzk4Nzg2Mzc2MzY0Y2QyNWUzNTMxNTg1NDM5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.WAFhGiOvEMNx18Ra66KQC6UZ_9FhyYtPwlxdrE8zMZU)

https://wiki.archlinux.org/title/Laptop/Apple#iMac_2020

bigfoot7375, 2024-12-31:
> https://wiki.archlinux.org/title/Laptop/Apple#iMac_2020

Didn't work, but few things that happened afterwards:
When it did boot, resolution was still 5k although I tried video=1920x1080 or video=2560x1440.
After returning to macOS, iMac automatically powred off after about a minute and when powering back on, got the "System was shutdown unexpectedly" message.
And the following message appeared on terminal (different than before) when was stuck:
[image]

AdityaGarg8, 2024-12-31:
In settings, what resolution options are visible to you?

AdityaGarg8, 2024-12-31:
Also, did you add it correctly? What's `cat /proc/cmdline`

bigfoot7375, 2024-12-31:
I've got the standard complete list (From 1024x768 up to 5120x2880 including the 1920x1080 and 2560x1440 options).
cmdline is irrelevant yet as I add it to the kernel parameters at boot time on install boot from usb device.
But for some reason it ignores it and when it boots, I have it initialized at 5k.

AdityaGarg8, 2024-12-31:
> I've got the standard complete list (From 1024x768 up to 5120x2880 including the 1920x1080 and 2560x1440 options). cmdline is irrelevant 

How is cmdline irrelevant even if you add at boot time?

> yet as I add it to the kernel parameters at boot time on install boot from usb device. But for some reason it being ignored and when it boots, I have it initialized at 5k.

bigfoot7375, 2024-12-31:
Don't have Linux installed so there's no cmdline yet.
I'm adding the "video=..." while booting from usb device to live system.
System not booting so I don't have the /proc/cmdline available.

AdityaGarg8, 2024-12-31:
Try using safe graphics option

AdityaGarg8, 2024-12-31:
For installation, some people use this workaround, copied the message from the discord!

Hey, for the amdgpu driver issue for IMacs with t2 kernel and 5K display I did not find a solution. I will try to dig into the driver code when I have more time to understand what is actually happening. For now I use this solution:

amdgpu-monitor.service
```
[Unit]
Description=AMDGPU Driver Monitor
After=multi-user.target

[Service]
Type=simple
ExecStart=/usr/bin/amdgpu_monitor
Restart=always
RestartSec=10
StandardOutput=syslog
StandardError=syslog
SyslogIdentifier=amdgpu-monitor

[Install]
WantedBy=multi-user.target
```

amdgpu_monitor
```sh
#!/bin/bash

# Error pattern to search for
ERROR_PATTERN=".*\[drm:amdgpu_device_init \[amdgpu\]\] \*ERROR\* hw_init of IP block <smu> failed -62.*"

# Monitoring interval in seconds
INTERVAL=5

while true; do
    if journalctl -k -b | grep -qE "$ERROR_PATTERN"; then
        echo "Detected amdgpu driver error (hw_init of IP block <smu> failed)! Rebooting..."
        sudo reboot
    fi
    sleep $INTERVAL
done
```

It's a simple reboot when it fails to load the driver properly. Maybe it's not super useful for classic users, but when like me you need to manage 200+ t2 IMacs it's a very good temporary patch.

So if anyone else have this issue, that's what I'm using for now.

bigfoot7375, 2024-12-31:
Using nomodset is helpfull for about 80% of the time (it's booting 4 times out of 5).
But this is only for installation and it disable the HW acceleration, right (It uses software rendering)?
Other than that, I'm not sure that the provided script will be helpful as It's getting stuck at a pretty early stage of the boot process (not sure if services are running at this time).

AdityaGarg8, 2024-12-31:
I've provided you all possible workarounds. There is currently no fix available for this. You probably can read the link below to know more about what you are actually facing:

https://khronokernel.com/macos/2021/12/08/5K-UEFI.html

Probably file a bug report with AMD.

closed 2025-02-20 by AdityaGarg8