t2linux/T2-Ubuntu #144: ubuntu 24.04 waking up

corykantar, 2025-01-06:
Maybe this is a known issue, not sure.  I am using a 2019 intel macbook pro (32gb ram) install went great, but after the laptop goes to sleep overnight, in the morning the keyboard and trackpad do not wake it up.  the power button will wake it, but then the keyboard and trackpad no longer work until a reboot.  Is there a fix for this?

AdityaGarg8, 2025-01-06:
Suspend is broken on T2 Macs and no fix is available yet.

closed 2025-01-06 by AdityaGarg8

corykantar, 2025-01-06:
is the idea then to disable suspend?

AdityaGarg8, 2025-01-06:
Yeah disable it, probably just shut down the Mac when not in use.

corykantar, 2025-01-06:
does it work on any other distro?

AdityaGarg8, 2025-01-06:
> does it work on any other distro?

Works on Arch Linux and EOS, not so reliably though.

lemmyg, 2025-01-06:
Usual workaround is to disable suspend in the desktop power profile settings. So it doesn't accidentally suspend when the computer is idle.

corykantar, 2025-01-06:
think it will be fixed in the future?

AdityaGarg8, 2025-01-06:
> think it will be fixed in the future?

Chances are pretty low.

walles, 2025-08-06:
There's a long conversation about this at https://github.com/t2linux/T2-Debian-and-Ubuntu-Kernel/issues/53.