t2linux/T2-Ubuntu #149: applesmc write_smc_mmio failed and GPU fan not spinning on Linux (T2 Mac)

pejman-hkh, 2025-04-19:
I'm running Linux on a Mac device with T2 chip, and I'm encountering an issue with the applesmc module. The following message appears in `dmesg`:

```
applesmc APP0001:00: write_smc_mmio(11 F1Mn) failed: 134
```

Questions:

What does this error mean?

Is it safe to ignore or does it indicate a hardware interaction failure?

Additionally:

Here is the output of sensors:
```
amdgpu-pci-0300
Adapter: PCI adapter
vddgfx:      712.00 mV 
fan1:           0 RPM  (min =    0 RPM, max = 4950 RPM)
edge:         +61.0°C  (crit = +110.0°C, hyst = -273.1°C)
                       (emerg = +115.0°C)
junction:     +63.0°C  (crit = +105.0°C, hyst = -273.1°C)
                       (emerg = +110.0°C)
mem:           +0.0°C  (crit = +150.0°C, hyst = -273.1°C)
                       (emerg = +155.0°C)
PPT:          15.00 W  (cap =  50.00 W)

applesmc-acpi-0
Adapter: ACPI interface
fan1:        3337 RPM  (min = 1836 RPM, max = 5616 RPM)
fan2:        3095 RPM  (min = 1700 RPM, max = 5200 RPM)
TA0V:         +22.2°C  
TB0T:         +33.8°C  
TB1T:         +31.8°C  
TB2T:         +33.8°C  
TC0E:         +58.0°C  
TC0F:         +58.8°C  
TC0P:         +53.0°C  
TC0T:          +0.0°C  
TC1C:         +56.0°C  
TC2C:         +55.0°C  
TC3C:         +54.0°C  
TC4C:         +56.0°C  
TC5C:         +56.0°C  
TC6C:         +56.0°C  
TC7C:         +57.0°C  
TC8C:         +56.0°C  
TCGC:         +58.0°C  
TCMX:         +58.0°C  
TCSA:         +55.0°C  
TCXC:         +57.2°C  
TF0S:          +5.8°C  
TG0P:         +57.0°C  
TG1P:         +56.2°C  
TGDD:         +64.0°C  
TGDE:         +55.0°C  
TGDF:         +64.2°C  
TGDT:          +0.0°C  
TGVF:         +61.0°C  
TGVP:         +61.0°C  
TH0F:         +40.5°C  
TH0X:         +40.8°C  
TH0a:         +37.8°C  
TH0b:         +40.2°C  
TH1a:         +35.8°C  
TH1b:         +40.8°C  
TM0P:         +52.0°C  
TPCD:         +60.0°C  
TTLD:         +51.0°C  
TTRD:         +51.5°C  
TW0P:         +51.0°C  
TaLC:         +45.5°C  
TaRC:         +45.2°C  
Th1H:         +61.2°C  
Th2H:         +51.5°C  
Tm0P:         +55.5°C  
Ts0P:         +32.8°C  
Ts0S:         +42.8°C  
Ts1P:         +28.8°C  
Ts1S:         +41.5°C  
Ts2S:         +42.8°C  

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +63.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +57.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +57.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +56.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +57.0°C  (high = +100.0°C, crit = +100.0°C)
Core 4:        +58.0°C  (high = +100.0°C, crit = +100.0°C)
Core 5:        +57.0°C  (high = +100.0°C, crit = +100.0°C)
Core 6:        +64.0°C  (high = +100.0°C, crit = +100.0°C)
Core 7:        +57.0°C  (high = +100.0°C, crit = +100.0°C)

BAT0-acpi-0
Adapter: ACPI interface
in0:          11.84 V  
temp:         +34.1°C  
curr1:         3.88 A  (avg =  +3.80 A)

pch_cannonlake-virtual-0
Adapter: Virtual device
temp1:        +60.0°C  

nvme-pci-0400
Adapter: PCI adapter
Composite:    +35.9°C
```

The fan for the GPU (amdgpu-pci-0300) always reports 0 RPM, while the other fans (from applesmc) are spinning properly.

Is this expected behavior?

Could this indicate that the GPU fan is not being controlled properly under Linux?

Is it managed by macOS firmware or should amdgpu be able to spin it under Linux?

AdityaGarg8, 2025-04-19:
`sudo rm /etc/udev/rules.d/30-amdgpu-pm.rules` try running this for gpu

for applesmc, ignore

pejman-hkh, 2025-04-19:
The file /etc/udev/rules.d/30-amdgpu-pm.rules does not exist on my system.

AdityaGarg8, 2025-04-19:
What ISO have you used? And what kernel version are you using?

pejman-hkh, 2025-04-19:
I installed Ubuntu 24.04 and then installed linux-t2 using apt install linux-t2.

My current kernel version is: 6.14.1-1-t2-noble.

AdityaGarg8, 2025-04-19:
Have you added the necessary kernel parameters?

pejman-hkh, 2025-04-19:
I haven't added any custom kernel parameters.  
I just installed it using `apt install linux-t2`.

AdityaGarg8, 2025-04-19:
https://wiki.t2linux.org/distributions/debian/installation/

Please follow any step you skipped from here.

Also, try enabling low power mode for AMDGPU:

```bash
cat <<EOF > /etc/udev/rules.d/30-amdgpu-pm.rules
KERNEL=="card[012]", SUBSYSTEM=="drm", DRIVERS=="amdgpu", ATTR{device/power_dpm_force_performance_level}="low"
EOF
```

pejman-hkh, 2025-04-19:
I have added the GPU settings, but the memory clock is still stuck at 100%.  
 I also added the `intel_iommu=on iommu=pt pcie_ports=native` parameters.

AdityaGarg8, 2025-04-19:
https://gitlab.freedesktop.org/drm/amd/-/issues

Please open to bug report with AMD. We don't really manage the GPU drivers.

closed 2025-04-19 by AdityaGarg8

pejman-hkh, 2025-04-19:
Is it normal that the GPU fan always shows 0 RPM in sensors output?

AdityaGarg8, 2025-04-19:
> Is it normal that the GPU fan always shows 0 RPM in sensors output?

Not many people care about this, but I assume it to be normal since people who have checked it are getting them as 0

pejman-hkh, 2025-04-20:
When the system goes to sleep, everything stops working afterward.
Do you have any solution for this issue?

AdityaGarg8, 2025-04-20:
> When the system goes to sleep, everything stops working afterward. Do you have any solution for this issue?

Sleep is broken. No solution yet.