t2linux/T2-Ubuntu #151: Everything working but not touchbar??

Maki9009, 2025-05-24:
So when i first did the install everything was working even the touch bar, after i restarted my mac. touchbar no longer works. its just turned off, Heres the thing though.. i kinda want to make my touchbar to display my battery life if thats possible please. since i made my laptop into a headless mac to run servers and cloud drives on it. i just want to have the ability to see its battery life or charging. since its always connected via thunderbolt type c... and i just have no idea if its charging or not unless i plug in the hdmi, and sometimes i want to unplug the charger since atm its my only charger.. but i just think it would be cool to have that battery life displayed on the touch bar.

Maki9009, 2025-05-24:
i did try to follow the touch bar custom setup you guys had but nothing really happened..

AdityaGarg8, 2025-05-24:
And if you need battery widget, open an issue here: https://github.com/AsahiLinux/tiny-dfr