t2linux/T2-Ubuntu #152: Freeze at Can not mount /dev/loop0 (/cdrom/casper/filesystem.squashfs) on /filesystem.squashfs

j0hnVu, 2025-06-03:
Macbook Air 2019
Secureboot Disabled

Select the option "Try or Install Kubuntu" results in:

(initramfs) stdin: Invalid argument
48+0 records in
48+0 records out
Failed to add #2 partition: Invalid argument
losetup: /cdrom/casper/filesystem.squashfs: failed to set up loop device: Invalid argument
stdin: Not a type writer
mount: mounting /dev/loop0 on /filesystem.squashfs failed: No Such Device
Can not mount /dev/loop0 (/cdrom/casper/filesystem.squashfs) on /filesystem.squashfs

AdityaGarg8, 2025-06-03:
What ISO burning software have you used?

j0hnVu, 2025-06-03:
Thank you for your reply. I used dd as instructed in your guide.
My dd cmd: 
sudo dd bs=4M if=kubuntu*.iso of=/dev/disk2 conv=sync status=progress

AdityaGarg8, 2025-06-03:
Use balena etcher. DD has been quite unreliable for some reason.

j0hnVu, 2025-06-03:
Booting ubuntu ISO imaged using dd with no conv=sync and running sync later was a success. I'll test with kubuntu image later. Thank you for your reply and your dedication to maintaining the project. I wish you the best.

closed 2025-06-03 by j0hnVu

gauthamrmn, 2025-06-12:
I had the same error and tried your solution with the Kubuntu Image, but it didn't work. Additionally, when I tried to use balena etcher, it said that the iso I got from the iso.sh script lacked a partition table and thus wouldn't be bootable.

AdityaGarg8, 2025-06-12:
> I had the same error and tried your solution with the Kubuntu Image, but it didn't work. Additionally, when I tried to use balena etcher, it said that the iso I got from the iso.sh script lacked a partition table and thus wouldn't be bootable.

Just ignore that balena etcher warning and continue.

AdityaGarg8, 2025-06-12:
Ok so the latest release turned out to be defective. I have removed it. It should be working now.