t2linux/T2-Ubuntu #153: Mac Pro 7,1 Builtin Speaker No Audio

paulscalise1, 2025-07-02:
Hello,

I have been successfully running Ubuntu using the t2linux kernel for quite some time now... thank you!!

But i have recently needed my internal speaker to work on my machine (2019 Mac Pro).
I ran the install audio config script seen in the wiki [here](https://wiki.t2linux.org/guides/audio-config/) without apparent failure but still do not see an audio output in settings.

I have restarted pipewire and rebooted to no avail.

Thank you in advance for any suggestions.

-Paul

AdityaGarg8, 2025-07-02:
Please share `uname -r` and `pactl info`

AdityaGarg8, 2025-07-04:
This is an extremely old kernel. Please upgrade your kernel.

https://github.com/t2linux/T2-Debian-and-Ubuntu-Kernel?tab=readme-ov-file#installation

paulscalise1, 2025-07-09:
I have updated my kernel to version ```6.12.36-1-t2-jammy``` and reloaded the audio configs from the wiki. After a reboot, the internal speaker interface still will not appear.

AdityaGarg8, 2025-07-09:
```
sudo apt install --reinstall apple-t2-audio-config
sudo apt purge apple-t2-audio-config
```

Try running this and restart

closed 2025-07-09 by AdityaGarg8