t2linux/T2-Ubuntu #159: BelenaEtcher says that the ISO is not bootable and can't boot after burning

fr33s7yl0r, 2025-09-17:
I downloaded Kubuntu 25.04 using the iso.sh but when I tried burning it on my USB drive, belenaEtcher warned that it was not a bootable ISO. I burned it anyway but then it would not pop up as a bootable option on the old macbook at all. Any ideas?

AdityaGarg8, 2025-09-17:
"old MacBook"? Are you referring to a T2 Macs here or non T2 one.

fr33s7yl0r, 2025-09-17:
Yes of course I am referring to the T2 macbook (Intel mid 2017 15", touch bar). Regardless of the version, a bootable USB should show up if the secure boot is turned off and I reboot into a boot menu with the CMD pressed during restart. Am I doing something wrong? 

The main question is: why would the ISO image be non-bootable?

AdityaGarg8, 2025-09-17:
Intel mid 2017 15", touch bar is NOT a T2 MacBook.

fr33s7yl0r, 2025-09-17:
the reason was to have the keyboard, mouse and wifi working during installation. I just tried burning a regular kubuntu ISO and I was able to boot without a single issue but of course the touchpad and keyboard don't work at all

fr33s7yl0r, 2025-09-17:
I just verified, the image provided here is NOT bootable. Even if it is burned using DD instead of belenaEtcher, I can't boot from it on any computer. In contrary, the image from the Kubuntu website booted without any issues

AdityaGarg8, 2025-09-17:
Hmm, ok. I'll see what's wrong. You can try using an older image for the time being.

fr33s7yl0r, 2025-09-17:
There is no macos on that laptop anymore. But I think I just mentioned that the image is not bootable by itself, regardless of the computer you are trying to boot it at. So what possible connection does it habe to the model identifier? 
If you need to know the model I can tell you without that: it is A1707

AdityaGarg8, 2025-09-17:
Yeah this is a non T2 Mac. The cause of your issue is most likely a regression that happened upstream. Can you check an official Kubuntu 24.04 ISO? I think keyboard works there

fr33s7yl0r, 2025-09-17:
The official image works, I installed Kubuntu but none of the drivers are there, including keyboard, trackpad, and Wi-Fi. It shows the network networks, but you cannot connect to any of them.

AdityaGarg8, 2025-09-20:
I am able to successfully boot the latest Kubuntu 25.04 ISO on a T2 Mac by burning using balena etcher. Closing this as completed.

closed 2025-09-20 by AdityaGarg8